jibrilsharafi/EnergyMe-Home #11: Question: What are the costs?

CommanderRedYT, 2024-10-26:
Hi! This is a really cool project. I was wondering how much the whole unit costs?

(Sorry for opening a issue, but I can't see the discussions tab)

jibrilsharafi, 2024-10-26:
Hi!

The unit cost (without the clamps) is around € 30, depending on where you source your PCB or components. The clamps then are about € 3 each.

You are right: I did not explain correctly how to reach 17 channels but I published everything in a rush. It has two expansion boards that can be placed on top of the main PCB. They are connected through standard header pins at the bottom and top of the PCB. In the documentation folder you will find all the schematics!

CommanderRedYT, 2024-10-26:
Ah, makes sense! 30€ is really cheap, thats really cool! Thanks for the cool project!

CommanderRedYT, 2024-10-26:
Hi, I have another question: How would one deal with three phases? Do I have to install one per Phase?

jibrilsharafi, 2024-10-27:
Yes, you would need to install one per phase.

CommanderRedYT, 2024-10-27:
Do you think it would be possible to contribute to the PCB and maybe add another two phases? Is it possible to have the easyeda archive for importing in kicad? (I haven't really used easyeda so I am not sure if they have something for collaboration)

jibrilsharafi, 2024-10-27:
Monitoring three-phase circuits would require many structural changes, including:
- changing the energy meter IC
- adding multiplexers
- replacing the 3.5mm jack with something smaller (otherwise the PCB will get very wide very quickly)
- re-design the high-voltage part to handle three-phase 400 V
- changing many things in the firmware as it was created to handle single-phase measurements
- the total costs will increase
Given this, I think it would be wiser to start another similar project based off this that is capable of working with three-phase instead.

You are right: in the documentation I forgot to upload the .zip containing the data about the project. Unfortunately I have never worked with KiCad so I will leave the EasyEDA backup here, hoping that you will manage to export it: [BackupProjects_jabrillo_personal_0_20241027.zip](https://github.com/user-attachments/files/17533610/BackupProjects_jabrillo_personal_0_20241027.zip).

jibrilsharafi, 2024-10-27:
I mean, now that I think about it, it is not impossible to monitor three phases. 
Of course you would need to change the code and do some approximation since you would only monitor one voltage. But by assuming the same voltage and power factor, and multiplying by the current, you could have the same hardware with 17 channels to monitor up to 5 full three-phase circuits. 
If you think it would be enough like this, I could work on my ideal solution.

CommanderRedYT, 2024-10-27:
Would be interesting if chips like the 78M6631 could work. The IC needs voltage and current, right? Like, it's not possible to assign the phase in software, right? With this IC you would need to have current sensor groups for L1, L2 and L3.

I mean around here the only place where people need three phase power is electric car chargers and electric stoves. Other than that, the three phases are distributed to different outlets so that they all can be the same amount.

Only problem is when you have a single-phase pv Installation (with microinverters for example) or a single-phase battery. So you could have a huge unbalanced current draw there.

Also, zero-feed-in would be interesting to implement with three separate devices.

jibrilsharafi, 2024-10-27:
I never used the 78M6631, but I guess it would work (ofc you would still need to change everything in the PCB and implement the communication specifics).

The ADE7953 which is currently used needs voltage and current. So you would need to do the assignment in the software of the ESP32 itself. 

I would do it like this: in the channel configuration page (below)
[image]
the user should have another dropdown menu indicating the phase. Now, depending on the chosen phase, the data would be read or computed. The approximation would require using the same power factor as the main line and the same voltage but multiplied by the current of the given channel.

Does this make sense to you?

CommanderRedYT, 2024-10-27:
Okay, so in case one would use the IC three times, one would need to have sensor input groups depending on which phase.

Same power factor would work as long as you do not have a huge consumer/producer  on only one phase I think.

jibrilsharafi, 2024-10-28:
The issue with 3 ICs is that you need to have static phase inputs (for instance, all channels from 0 to 6 are phase A, 7 to 14 are phase B, etc.). This would leave less flexibility while still requiring a complete PCB change.
I will probably do some tests with the current hardware to see if reading 3-phase power is still feasible with these assumptions.

jibrilsharafi, 2024-11-07:
In the end I managed to make the three phase work in the latest pull request #12 !

jibrilsharafi, 2025-02-03:
Hi @bsaggy!
Thanks for your interest in the project! Let me address your questions:

About pick & place files: I didn't upload them for the TOP and BOTTOM boards since they have very few components. However, I can generate and upload them if you'd like to use assembly services for those boards too.

Regarding PCB manufacturing: I've always used JLCPCB myself. While I did get the first prototype assembled by them, I've since found that hand-soldering these components is quite manageable and gives me more flexibility (plus significant cost savings). This might be a route worth considering if you're comfortable trying some SMD soldering (you will need a reflow gun).

For the SCT-013 sensors: I source them from Amazon or AliExpress (I'm based in Italy). Both platforms usually have competitive prices for bulk orders.

Flashing process: You'll need to connect a standard UART programmer to the programming header on the board (which I apologize for not labelling in v4!). Connect 3V3, GND, TX, and RX, then you can either read the serial output or program it by holding both the program and reset buttons together. There's plenty of documentation online about ESP programming if you need more details.

Quick update: I'm currently **testing v5** (hopefully the **final version**), which includes some improvements like 4-layer design and optimized component placement. I'll be uploading comprehensive documentation to the development branch soon.

Feel free to open any issues if you have more questions - happy to help!

jibrilsharafi, 2025-02-03:
You're welcome @bsaggy!

Sorry, but I'd rather not go through the hassle of restoring a previous version and creating the pick-and-place files without proper revision. 
In any case, I advise you to learn to solder SMD components: it will definitively allow you to step up your electronics design capabilities! You only need a hot gun, a hot plate, and solder paste.

**The current sensors MUST have an output of 1V** or less (some are 500 or 333 mV), but surely not in current. My design does not use burden resistors to further optimize components and space.

The design of v4 is already stable and tested for many months, so feel free to use that if you are in a rush. Otherwise, I will surely release the new version in the following weeks.

closed 2025-02-07 by jibrilsharafi